TelluIoT/ThingML #73: Order of instanciation is random or follow an implicit rule

Lyadis, 2015-03-12:
The order of instanciation seems to be backward or random.

Why this could be a problem?
-> In arduino if you initialize the Serial port in one instance (at the begining) and use it in an other (in the begining) you're not sure it will be initialized before you use it.

It could be usefull to be able to decide it explicitly.

Here are some examples which seems to follow a backward rule, but in more complexe examples I can't say if it still follows this rule or not.

------ Config ------

```
configuration PingPongCfg
@arduino_stdout "Serial"
{
    instance client1: Client1
    instance client2: Client2
    instance server: Server
    connector client1.myPort => server.myPort
    connector client2.myPort => server.myPort
}
```

------ Result ------

// The Init messages are in the "on entry" of thing's statechart

[ Server ] Init server
[Client 2] Init Client 2
[Client 1] Init Client 1
[Client 1] Send Ping
[ Server ] Ping received, sending pong
[Client 1] Pong received
[Client 2] Pong received

------ An other exemple ------ 

```
configuration PingPongCfg
@arduino_stdout "Serial"
{
   
    instance server: Server <==================
    instance client1: Client1
    instance client2: Client2
    connector client1.myPort => server.myPort
    connector client2.myPort => server.myPort

}
```

------ Result ------

[Client 2] Init Client 2
[Client 1] Init Client 1
[Client 1] Send Ping
[ Server ] Init server <==========================
[ Server ] Ping received, sending pong
[Client 1] Pong received
[Client 2] Pong received

brice-morin, 2015-03-12:
Thatt is the magic of EMF.... The order on collections is... magic...

brice-morin, 2015-10-28:
I am currently implementing some helpers that will return the list of dependents clients/servers for a given instances, etc. So that, at least in some cases, we can have a start order that respects client/server dependencies (typically works if it is no cycles in the connectors).

brice-morin, 2015-10-28:
OK, I added derive properties in configuration and implemented a [proper ordering in the JS compiler](https://github.com/SINTEF-9012/ThingML/blob/master/compilers/javascript/src/main/java/org/thingml/compilers/javascript/JSCfgMainGenerator.java#L227), which you (@Lyadis) can port to the C compiler. Basically we start by all instances without any server, then we continue with instances clients of the set of instances of the previous set, etc. At the end (typically if it is some cycles, we start all the rest in whatever order).

I would need to test a bit more on some tricky cases, but it is at least a starting point, which should solve most of the ordering problems.

brice-morin, 2015-10-28:
> you should not use `List.contains` or `List.containsAll` as the `equals` in EMF is broken (**!!!**). Rather use the `contains` and `containsAll` methods I have defined, which use `EcoreUtil.equals`

Lyadis, 2015-11-09:
I implemented something in the C compiler, and added some test.

Some tests appear not to be working with Java and JS compiler, but whether or not it is a problem of specification or implementation can be discussed.

The configuration to be tested is the following:

```
connector B.r => A.p
connector C.r => B.p
connector B.r => D.p
connector D.r => C.p
connector E.r => C.p
```
where A,B,C,D,E are instances, r is a required port, and p a provided one.
[image]

The order produced with JS compiler is ADBEC while in my opinion E should be initialized in last, because it requires C.

brice-morin, 2015-11-09:
Yes, seems like a bug

brice-morin, 2015-11-09:
Well, since you got the algorithm right (I am preparing a complex test, hehe) and since it is good for you to get started with Java and JS compiler, I re-assigned the issue to you, so that you can fix it in those compilers. :-)

commit 460cbc72 (2015-11-10): fix ordering issue in Java and JS (use same algorithm than for C). Fixes #73
closed 2015-11-10 by brice-morin