TelluIoT/ThingML #77: JS error with thing name in lower case

lmouline, 2015-04-17:
```
thing test {}

configuration testCfg {
    instance t : test    
}
```

JS error : 

```
module.exports = test;
                 ^
ReferenceError: test is not defined

```

closed 2015-04-19 by brice-morin

lmouline, 2015-04-21:
I have two bugs now (with the same ThingML code)

```
module.js:338
    throw err;

Error: Cannot find module './Test'
```

So I change the file name : 
```
mv test.js Test.js
```

It works. 
But, when I shutdown, I have another bug : 

```
/tmp/ThingML_temp/testCfg/main.js:9
testCfg_t._stop();
          ^
TypeError: undefined is not a function
```

I try to fix it.

lmouline, 2015-04-21:
I cannot commit.

My modification : 

In ```org.thingml.compilers/src/main/scala/org/thingml/jsgenerator/JavaScriptGenerator.scala``` l.200 :
```
thing.generateJavaScript(ctx.getBuilder(self.getName + "/" + ctx.firstToUpper(thing.getName) + ".js"), ctx)
``` 

In ```org.thingml.compilers/src/main/java/org/thingml/compilers/main/JSMainGenerator.java```  l.146 :
```
 for (Instance i : cfg.allInstances()) {
            if(i.getType().allStateMachines().size() > 0) {
                builder.append(i.getName() + "._stop();\n");
            }
        }
```

brice-morin, 2015-04-21:
> Envoy� : mardi 21 avril 2015 09:23
> � : SINTEF-9012/ThingML

Ask @ffleurey to give you the rights on the main ThingML repo.

ffleurey, 2015-04-21:
Done.

brice-morin, 2015-04-21:
Good good :-)

commit 0b4d3b03 (2015-04-19): fix #77