TelluIoT/ThingML #80: Java import in ThingML (Path problem)

Lyadis, 2015-05-11:
I have this project which contains:

- JavaLib/myClass.java
- Test.thingml 

```
datatype myClass
	@java_type "JavaLib.myClass";

thing Test {
	property myA : myClass = 'new JavaLib.myClass('& 3 &');'
	
	statechart TestChart init Init {
		state Init {
			on entry do
				print "[Test] Start"
				''&myA &'.displayA();'
				print "[Test] End"
			end
		}
	}
}

configuration TestCfg {
	instance t : Test
}
```

How do I specify the path of myClass. Because so far, the only way I made it works is puting it in the folder thingml-gen/java/src/main/java or with an absolute path.

brice-morin, 2015-05-11:
Package your lib in a Maven project that you install locally.

Then in your thing, use the `@maven_dep "JavaLib"` annotation to tell the thing where to find the dependency.

Try that and close the issue if it works :-)

Lyadis, 2015-05-11:
@maven_dep seems to copy directly its content in the pom.xml between <dependecies></dependecies>

so I need to write it this way, to make it work. Is it the proper way?
```
@maven_dep "

<dependency>
    <groupId>MyGroupeID</groupId>
    <artifactId>MyArtefactID</artifactId>
    <version>0.0.1-SNAPSHOT</version>
</dependency>"
```

brice-morin, 2015-05-11:
yes indeed

closed 2015-05-11 by Lyadis

lmouline, 2015-05-11:
Is it possible to have a syntaxe like : 
@maven_dep "groupID:ArtefactID:Version" ?

And parse it.

brice-morin, 2015-05-11:
Yes, you can implement that feature, in addition (not in place) of the current one. :-)