TelluIoT/ThingML #81: JS - Error with "complex" arithmetic expression

lmouline, 2015-05-12:
When in ThingML I declare : 
```
var v1 : Integer = 4
var v2 : Integer = 3
var v3 : Integer = (v1 + v2) * ((v1 - v2) * 2)
```

The JS generated code is : 
```
var v1__var = 4;
var v2__var = 3;
var v3__var = v1__var + v2__var * v1__var - v2__var * 2;
```

The parentheses are missing.

brice-morin, 2015-05-12:
I will create a test case for that, and check the JS, Java and C compiler to see if the problem is just in the JS compiler or if it is a problem with ThingML parser.

However, @ffleurey broke ThingML and I cannot use it right now (but I will test on a version just before his commit...)

brice-morin, 2015-05-12:
OK, it should be fixed. We did not generate parenthesis in Java and JS

commit 593f3df4 (2015-05-12): fix #81
closed 2015-05-12 by brice-morin