TelluIoT/ThingML #82: java_type in JS compiler

lmouline, 2015-06-16:
In the JSMainGenerator class, there is : 

```
public static String getDefaultValue(Type type) {
        if (type.isDefined("java_type", "boolean"))
            return "false";
        else if (type.isDefined("java_type", "int"))
            return "0";
        else if (type.isDefined("java_type", "long"))
            return "0";
        else if (type.isDefined("java_type", "float"))
            return "0.0f";
        else if (type.isDefined("java_type", "double"))
            return "0.0d";
        else if (type.isDefined("java_type", "byte"))
            return "0";
        else if (type.isDefined("java_type", "short"))
            return "0";
        else if (type.isDefined("java_type", "char"))
            return "'\u0000'";
        else
            return "null";
    }
```

Is it supposed to be this way? It should be "js_type", right?

brice-morin, 2015-06-16:
Well, in principle you are right. In practice, the notion of type in JS is somehow a joke.... And I guess most Java type we use are the same in JS.

lmouline, 2015-06-16:
OK Ok.
So when we use ThingML for a JS platform and we have to declare a new type we have to do something like this : 

```
datatype X
    @java_type "Y";
```

I think it's a weird. In my mind,  why you don't keep the js_type annotation?

brice-morin, 2015-06-16:
Yes, we should indeed. But it is not a big deal. You can fix it :-)

lmouline, 2015-06-16:
ok ^^
I add this to my todo list ^^
But before I must fix the merge XD

closed 2015-06-17 by lmouline