TelluIoT/ThingML #86: JS helloworld leave the Bye state

barais, 2015-07-15:
In the helloworld sample,the js version of helloworld leave the bye State

lmouline, 2015-07-15:
With this statechart : 
```
statechart HelloWorld init Greetings {
		state Greetings {
			transition -> Bye 
			action print "Hello World!\n"
		}
	
		state Bye {
			on entry print "Bye.\n"
                        on exit print "exit"
		}
	}
```

I have this output (in java/js/posix): 
```
Hello World!

Bye.

```

Maybe the bug was fixed with the new version of the compiler

brice-morin, 2015-07-15:
Well. I fixed that "bug" but reverted it. That is the semantics of Node.js. When it is nothing more to do, it stops. Not that we do not exit the bye state (we enter it). But it is a sink state, hence nothing more will ever happen and node.js process stops. Make a simple node.js program that just print a line and it will stop the same way.

closed 2016-04-28 by Lyadis