TelluIoT/ThingML #92: Java generated code use old Jasm Version

lmouline, 2015-08-07:
JASM has been updated a few days ago. However, the Java generated code still use the old version of the library (and the version on the maven repository is still the old version).

Is it possible to update the Java Compiler and the JASM version on the maven repository?

brice-morin, 2015-08-07:
Hmmm... I thought I updated the compiler. Do you still have 'new Connector' in the generated Main file?

lmouline, 2015-08-07:
I am going to check.
But, in the generated code the Things still call the Component.send method. But, its doesn't exist anymore.

lmouline, 2015-08-07:
Ok the problem becomes from my computer (but not from me ^^)
I have some black magic things  that caused the disappearance of some methods in the Port class...

closed 2015-08-07 by lmouline

brice-morin, 2015-08-07:
Should I call an exorcist?