TelluIoT/ThingML #101: Broken JS tests

Lyadis, 2015-11-26:
I think that since the default output directory of the JS compiler is not the name of the configuration, tests are broken...

By the way, is the change of the default directory something we want to do with the c compiler too?

brice-morin, 2015-11-26:
Yes, I changed the JS compiler default output directory, because it generated in `/tmp/ThingML_temp/cfgname/cfgname` (at least from the HEADS IDE and standalone editor). We should change the way the test framework calls that compiler.

brice-morin, 2015-11-26:
For some reason, my virtual box is not working (I need Linux to run the tests). But I guess you can play [around this line](https://github.com/SINTEF-9012/ThingML/blob/master/org.thingml.tests/src/test/resources/execute.py#L116) to fix the problem by passing the proper params to the command line compiler

Lyadis, 2015-11-26:
Yes I fixed it, but some test with CEP are not working anymore. It seems to be an issue with the installation of reactivex...
```
module.js:338
    throw err;
          ^
Error: Cannot find module 'rx'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/home/sintef/Documents/ThingML/org.thingml.tests/target/tmp/_javascript/TestStreams/TestStreams.js:2:10)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
```

brice-morin, 2015-11-26:
that is strange... is the rx dependency declared in package.json?

Lyadis, 2015-11-26:
It does'nt seem so.

closed 2015-12-11 by Lyadis