TelluIoT/ThingML #102: Empty Internal transition

Lyadis, 2015-11-26:
Is this code allowed in ThingML?

```
thing TestEmptyInternalTransition
{
	statechart TestEmptyInternalTransition init I {
		property i : UInt16 = 0
	
		state S1 {
			internal
			guard i == 0
			action do
				print "Internal Empty transition"
				i = i + 1
			end
			
			transition -> S2 guard not(i == 0)
		}

		state S2 {
			on entry print "Done\n"
		}
	}
}
```
Because it work with the C compiler (State S2 is reached) but not with JS compiler.

brice-morin, 2015-11-26:
Yes, I guess it should be allowed. But not allowed (or discouraged if you do not have a guard, as it will loop forever).

Lyadis, 2015-11-26:
In my opinion a warning is preferable to an error, as it can be the intended behaviour.

brice-morin, 2015-11-26:
As for Javascript, I have to investigate a bit more, but it might be an issue in the JS library we are using, and we might need to contact the maintainer of that repo (who is typically quite responsive)

brice-morin, 2016-04-28:
Confirmed issue in state.js library, see steelbreeze/state.js#22

commit 6785f05e (2016-05-04): fix #102
closed 2016-05-04 by brice-morin