TelluIoT/ThingML #107: Group support

Lyadis, 2016-01-14:
Do we still want to support Configuration nested inside another configuration?

```

configuration NestedCfg
{
    ...
}

configuration Cfg
{
    group g : NestedCfg
    ...
}
```
Because it is still at the root of some bugs (See #26), and it makes both the ThingML and generated  code uglier and mroe confusing.

Lyadis, 2016-01-14:
@ffleurey @brice-morin ?

brice-morin, 2016-01-14:
I agree `group`s have many downsides. And I do not think they are used by our partners.

ffleurey, 2016-01-14:
Agree as well.

Lyadis, 2016-01-14:
Do you know who used the group notion? :-D ... Tests!!

brice-morin, 2016-01-14:
Hehe... :-)

Should be easy to fix though

brice-morin, 2016-01-15:
You mean the C tests, that use an old version of timer, that indeed relies on a configuration fragment to wrap the clock, etc.

Lyadis, 2016-01-15:
Yes. But it is indeed an smal fix.

commit 711aa7e9 (2016-01-14): fix #107 + added new advanced features like <=, >=, ++ and --
closed 2016-01-15 by Lyadis