TelluIoT/ThingML #108: Missing keyword "event" leads to erronous code

sdalgard, 2016-01-21:
When working implementing timer as a network plugin I found a strange behaviour.

When leaving out the keyword event in a transition like this:
	transition -> Init /*event */  timer?timer_timeout
	action do
		'printf("%d %s", GetOsTimeMs(), ' & name & ');'
	end
	
 The additional code is generated in the loop() method:
ActiveTimer_handle_empty_event(&TestTimerSerial_client2a_var);
ActiveTimer_handle_empty_event(&TestTimerSerial_client0_var);
ActiveTimer_handle_empty_event(&TestTimerSerial_client1_var);

The attaced example has replaced are using the serial network plugin.
This is done to make the example compilable using a standard compiler build.
 
[errorcase.zip](https://github.com/SINTEF-9012/ThingML/files/98903/errorcase.zip)

brice-morin, 2016-01-21:
Well, it seems the editor in Eclipse reacts on that missing event keyword and reports and error. In principle, we should not generate any code. As we are still in a development phase, we do not block code generation if it is errors in models as it is sometimes useful to be able to force the compilation. But in the next few months, we will probably be more strict and refuse to compile erroneous models.

brice-morin, 2016-03-17:
The new version should not refuse to compile when the input file contains syntax error

closed 2016-03-17 by brice-morin