TelluIoT/ThingML #113: Update site does not keep history

brice-morin, 2016-03-07:
It would be good that the update site at least keeps the latest release in addition to the current snapshot. Right now, only the version corresponding to the latest commit will be in the update site. We can of course manually enable/disable the build of the update site, but it is not convenient.

skorsky, 2016-08-01:
this would be helpful. I had trouble restoring a previous version in connection with issue #129.

brice-morin, 2017-01-17:
When we make a release, the update site will be available on GitHub, into the [releases](https://github.com/SINTEF-9012/ThingML/releases). The other update site will contain the latest version of ThingML. In any case, it should now be possible to fallback to a stable update site.

closed 2017-01-17 by brice-morin