TelluIoT/ThingML #115: Error with String parameter in java compiler

Lyadis, 2016-04-19:
CustomPrint for java in HEADS tutorials does not compile anymore:

```
java.lang.ClassCastException: org.sintef.thingml.impl.ObjectTypeImpl cannot be cast to org.sintef.thingml.PrimitiveType
        at org.thingml.compilers.java.JavaThingImplCompiler.generateMessages(JavaThingImplCompiler.java:98)
        at org.thingml.compilers.java.JavaThingImplCompiler.generateImplementation(JavaThingImplCompiler.java:496)
        at org.thingml.compilers.java.JavaCompiler.do_call_compiler(JavaCompiler.java:106)
        at org.thingml.compilers.utils.OpaqueThingMLCompiler.compile(OpaqueThingMLCompiler.java:65)
        at org.thingml.compilers.commandline.Main.main(Main.java:192)
```

Lyadis, 2016-04-19:
It's probably caused by the on going work on serialization. JavaThingImplCompiler.java:98 seems to assume that parameter type are always datatype and never object, which is a requirement only for messages exchanged over network, but not for internal communications.

brice-morin, 2016-04-19:
Yes indeed. Should be easy to fix.

closed 2016-08-19 by Lyadis