TelluIoT/ThingML #117: Bad migration of a derived property

brice-morin, 2016-05-20:
```
[Error] java.lang.ClassCastException: org.sintef.thingml.impl.SessionImpl cannot be cast to org.sintef.thingml.ParallelRegion
[Error]         at org.sintef.thingml.helpers.ParallelRegionHelper.allContainedSessions(ParallelRegionHelper.java:41)
[Error]         at org.thingml.compilers.java.JavaThingImplCompiler.generateImplementation(JavaThingImplCompiler.java:585)
[Error]         at org.thingml.compilers.java.JavaCompiler.do_call_compiler(JavaCompiler.java:107)
[Error]         at org.thingml.compilers.utils.OpaqueThingMLCompiler.compile(OpaqueThingMLCompiler.java:65)
[Error]         at org.thingml.compilers.commandline.Main.main(Main.java:200)
```

brice-morin, 2016-05-27:
I guess someone else than @ffleurey fixed this, isn't it @Lyadis ?

Lyadis, 2016-05-27:
If I am guilty of this fix, it is non intentional...

closed 2016-05-27 by Lyadis