TelluIoT/ThingML #122: error after compiling an incomplete file

omidkhodatars, 2016-06-10:
There was an error in ThingML compiler, it happened after I tried to compile an incomplete file. 
The file could not be displayed anymore after that 

ERROR: Exeption calling ThingML Compiler: nullPlease contact the ThingML development teamjava.lang.StackOverflowError
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:125)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
[... 933 lines cut ...]
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)

brice-morin, 2016-06-10:
Send your model, so that we can reproduce the bug here. A stack trace (especially when it is only within code generated by EMF) is not something we can easily exploit.

omidkhodatars, 2016-06-10:
Error happened with ThingML version:0.7.0.201606071315	.
and Here is the incomplete code 

import "datatype.thingml"	
thing HighTemperatureDetection {
	// message definition
	message temperatureSensor (id : integer, temp : float);
	message temperatureSensorCEP (id : integer, temp : float, msg : string);
	message PressureSensor (id : integer, pressure2 : float);
	//port definition
	
	internal port eventPort { 
		sends , temperatureSensor, temperatureSensorCEP, PressureSensor
		receives , temperatureSensor, temperatureSensorCEP, PressureSensor
	}
	//find stream query
	// stream query conversion
	stream randomName
	from t : eventPort?temperatureSensor  :: keep if t.temp > 20.0 and t.id == 10 
	select 	
		  var nid:integer = t.id, 
		  var ntemp:float = t.temp, 
		  var nmsg:string = "CEP"
	produce eventPort!temperatureSensorCEP(nid, ntemp, nmsg)

Lyadis, 2016-06-10:
You have some thingml syntax issues:
 * remove comma between `sends` and  `temperatureSensor` same between `receives` and  `temperatureSensor` (in the port declaration)
 * remove comma after `var nid:integer = t.id` and the line below. (in the stream declaration)
 * add a configuration, and close the bracket of your thing (but I suppose that this fragment of code is not the complet file) (Likewise, your thing has no behaviour.)

omidkhodatars, 2016-06-10:
Yes, I have already known that the syntax is not correct in this auto-generated code , 
but the main problem is after compiling this code instead of just having a compile error , the ThingML IDE is not able to display the file anymore. 
[image]

brice-morin, 2016-06-10:
We will redo the parser this summer. Hopefully it will be more robust to that kind of things and will better report errors.

omidkhodatars, 2016-06-10:
I just wanted to report the error in order to improve the future implementation

brice-morin, 2017-06-22:
Hopefully, the new parser and checker are now OK

closed 2017-06-22 by brice-morin