TelluIoT/ThingML #124: CEP examples can not be compiled

omidkhodatars, 2016-06-11:
Hi, 
Apparently the new version of ThingML	0.7.0.201606071315	
does not work for CEP syntax, not working even for the examples provided in tutorials. 

it was working fine in the previous version ->  ThingML	0.7.0.201604281252

brice-morin, 2016-06-13:
Hi,
OK, I updated the plugins. Also fixed an issue in the samples in the tutorials (it was two streams with the same name, and that should not be allowed). Try and update the issue accordingly.
Thanks!

omidkhodatars, 2016-06-14:
I just update it but, there are still errors, a thingml file cannot be parsed and stack overflow error for almost all of them . see some of them in the snapshots . 
after the error displaying , IDE cannot display the thingml file with thingml editor anymore .
and after closing IDE it cannot be opened again because of the errors in the log file(see snapshot)

[image]

[image]
[image]

brice-morin, 2016-08-05:
Parsing of CEP should now be fixed. Re-open if you still have a problem.

closed 2016-08-05 by brice-morin