TelluIoT/ThingML #129: EMFText Editor does not open on .thingml file -StackOverflowError

skorsky, 2016-07-20:
I get the following error when I open a .thingml file with the EMFText thingml editor.
I use version 0.7.0.201606071315 of org.thingml.eclipse.feature.feature.group.

[datatype.thingml.txt](https://github.com/SINTEF-9012/ThingML/files/374143/datatype.thingml.txt)
[FilterExampleAutoGeneratedFile.thingml.txt](https://github.com/SINTEF-9012/ThingML/files/374144/FilterExampleAutoGeneratedFile.thingml.txt)

eclipse.buildId=4.6.0.I20160606-1100
java.version=1.8.0_102
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -product org.eclipse.platform.ide
Command-line arguments:  -product org.eclipse.platform.ide -data C:\Workspaces\Heads Transpiler/../runtime-EPLtoThingMLTranspiler -dev file:C:/Workspaces/Heads Transpiler/.metadata/.plugins/org.eclipse.pde.core/EPL to ThingML Transpiler/dev.properties -os win32 -ws win32 -arch x86_64 -consoleLog

org.eclipse.ui.workbench.texteditor
Error
Wed Jul 20 17:58:30 CEST 2016
Editor could not be initialized.

java.lang.StackOverflowError
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1081)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
[... 948 lines cut ...]
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)
	at org.sintef.thingml.impl.SimpleSourceImpl.eSet(SimpleSourceImpl.java:157)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1071)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eSet(BasicEObjectImpl.java:1095)
	at org.sintef.thingml.impl.ThingMLElementImpl.eSet(ThingMLElementImpl.java:130)
	at org.sintef.thingml.impl.AnnotatedElementImpl.eSet(AnnotatedElementImpl.java:133)
	at org.sintef.thingml.impl.SourceImpl.eSet(SourceImpl.java:131)

brice-morin, 2016-07-26:
OK, seems it was some problem with the pressure message that was sent but not defined:

```
datatype integer<2>
@js_type "int"
@java_type "int"
@java_primitive "true";
    
datatype boolean<1>
@js_type "bool"
@java_type "boolean"
@c_type "boolean";	

datatype float<4>
@java_type "float";

object string 
	@c_type "char *"
	@c_byte_size "2"
	@java_type "String"
	@js_type "String";

	
thing FilterExample {
	// auto-generated message definition
	message TemperatureSensor (id : integer, temperature : float);
	message PressureSensor (id : integer, pressure : float);
	message TemperatureSensorCEP (id : integer, temperature : float, msg : string);
	//auto-generated port definition
	
	internal port eventPort {
	sends  PressureSensor, TemperatureSensor, TemperatureSensorCEP
	
	receives  PressureSensor, TemperatureSensor, TemperatureSensorCEP
	}
	// auto-generated stream query 
	 
	stream randomName1hnm2aa75g2ja
	from t : eventPort?TemperatureSensor  :: keep if t.temperature > 35.0  
	select 	
	var newid:integer = 
	t.id, 
	var newtemperature:float = 
	t.temperature, 
	var newmsg:string = 
	"High Temperature Alarm"
	produce eventPort!TemperatureSensorCEP(newid, newtemperature, newmsg)
	
//auto-generated statechart-------------------------------
	statechart CEPstatechart init Base {      
		state Base {
		on entry do
		//input events simulation 
		eventPort!TemperatureSensor(10 , 51)
		eventPort!PressureSensor(10 , 51)
		eventPort!TemperatureSensor(11 , 52)
		eventPort!PressureSensor(10 , 52)
		eventPort!TemperatureSensor(12 , 15)
		eventPort!PressureSensor(10 , 15)
		eventPort!TemperatureSensor(13 , 53)
		eventPort!PressureSensor(10 , 53)	
		end
		 internal event ts: eventPort?TemperatureSensorCEP
		 action do
		print
		"Information of the filtered Event- id :"+
		ts.id+
		", temperature:"+
		ts.temperature+
		"message :"+
		ts.msg
						+ "\n "
		
		end
		}	
  	}
													
} // End of thing
		
		//auto-generated configuration 
		configuration FilterExampleCfg 
			{	
			instance my_instance: FilterExample
			//connector my_instance.receivePort => my_instance.sendPort	
			}

```

I will also re-generate the plugins so that you can update (but not sure it is necessary).

brice-morin, 2016-07-26:
Seems the problem is still there in Eclipse. Meanwhile, you can use the standalone editor, which seems to work on your example:

```
git clone https://github.com/SINTEF-9012/ThingML.git
cd ThingML
mvn clean install
cd org.thingml.editor.standalone
mvn exec:java
```

brice-morin, 2016-07-26:
Maybe someone did not re-generate the parser for Eclipse when updating the grammar. I will check tomorrow.

skorsky, 2016-08-01:
There is still no update of the Eclipse plug-ins which includes a fix for this.

brice-morin, 2016-08-01:
No, unfortunately, the plugins were up to date. Try the standalone editor (see above for instructions) for now, which does not seem to be affected by this issue.

@ffleurey will actively work on the new XText parser when he is back. Meanwhile, as a workaround exists, it does not seem too reasonable to invest too much effort in fixing the old parser

brice-morin, 2016-08-05:
Well, everything works in development mode when I run our plugins as Eclipse application... But when I update my HEADS IDE, it still does not work... Though it should be the very same code...

brice-morin, 2016-08-05:
Should now be fixed.

Note that there will be regressions in CEP for C/Arduino, as the annotations on CEP sources were crashing the whole parser.

closed 2016-08-05 by brice-morin

skorsky, 2016-08-05:
Is fixed now.

Thanks!

commit bc482f07 (2016-08-05): fix #129, fix HEADS-project/heads_ide#96