TelluIoT/ThingML #130: Local tests are broken

brice-morin, 2016-08-11:
It seems the local execution of the tests is broken. The ThingML files produced all contains error. Please fix ASAP

Lyadis, 2016-08-12:
I don't have the problem on my machine with the latest commit... What kind of errors are you dealing with?

brice-morin, 2016-08-12:
Seems like the problem is that it generate one absolute path with `\` on Windows, for one of the import, and that it does not parse correctly. Something like: `import "C:\Users\xxx\ThingML\testJar\src\main\resources\tests\StateMachine\testCompositeStates.thingml"`

brice-morin, 2016-08-12:
Something like `import "file://C:/Users/xxx/ThingML/testJar/src/main/resources/tests/StateMachine/testCompositeStates.thingml"` seems to work. Just try on Linux if `file://` works.

Lyadis, 2016-08-12:
No it doesn't seem to.

brice-morin, 2016-08-12:
OK, then `import "/C:/Users/xxx/ThingML/testJar/src/main/resources/tests/StateMachine/testCompositeStates.thingml"` seems to work on Windows. So it is basically to replace `\` with `/` and add a leading `/` if it is not there. That should be generic enough to work everywhere.

commit b9273625 (2016-08-12): Fixes #130
closed 2016-08-12 by Lyadis