TelluIoT/ThingML #132: Litteral 4294967295 is not parsed

Lyadis, 2016-10-24:
The following code isn't parsed. Meanwhile if `4294967295` is replaced by `'4294967295'`, it works.

```
datatype UInt32<4>	
	@c_type "uint32_t"
	@java_type "long"
	@java_primitive "true";

thing TestUInt32 {

	property min : UInt32 = 4294967295
    
    statechart TestType init S0 {
    
    	state S0 {
    		transition -> S1
			guard min == 4294967295
    	}
	state S1 {}
    }
}

configuration TestUInt32Cfg {
	instance tT : TestUInt32
}
```

brice-morin, 2017-02-09:
Relates to #156

jakhog, 2017-08-05:
I think this issue is to this line in the ThingML grammar ([here](https://github.com/TelluIoT/ThingML/blob/02b65d24b3314ac7b2e629746be7ea28c4619bba/language/thingml/src/org/thingml/xtext/ThingML.xtext#L407)):
```
terminal INT returns ecore::EInt: ('0'..'9')+;
```

It seems that the `ecore::EInt` type uses `int`s in Java. And while that sound fine at first, it is actually a problem - since the integer literal only represents the absolute value of the number (it doesn't include the `-` part).

This means that when you write `-2147483648` the parser actually tries to parse `2147483648` - which is outside the range of Java `int`s.

So - to fix this issue - I think we have to switch to `ecore::ELong`, and possibly check that the literal is within range somewhere else (either in a custom TerminalConverter, or in the Checker).

jakhog, 2017-08-05:
Since I'm not a certified Xtext guru yet - I'll give the responsibility back to the Franck guru

commit 685f38a1 (2018-01-29): Changed INT terminals from ecore::EInt to ecore::ELong

Should have no effect on generated platform code, as this is only used
internally in the ThingML compilers.

Fixes #132
closed 2018-01-29 by jakhog