TelluIoT/ThingML #133: org.thingml.eclipse.ui: FrameworkEvent ERROR in Eclipse

skorsky, 2016-10-31:
After Updates of ThingML in Eclipse, I get the error below.

The file C:\Tools\Eclipse4.6\configuration\org.eclipse.osgi\2416\0\.cp\lib\org.thingml.editor.standaloneApp-0.7.0-SNAPSHOT-jar-with-dependencies.jar is not a jar.
It contains this text (size 132 bytes):
version https://git-lfs.github.com/spec/v1
oid sha256:8d9ed40c01fc5f5f167338876af3dbe40ba758f4cb5fad9f2305510bf62c0079
size 9941884

Entry from Eclipse Error log:

eclipse.buildId=4.6.1.M20160907-1200
java.version=1.8.0_112
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -showlocation
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -showlocation

org.thingml.eclipse.ui
Error
Mon Oct 31 09:21:16 CET 2016
FrameworkEvent ERROR

java.util.zip.ZipException: Exception in opening zip file: C:\Tools\Eclipse4.6\configuration\org.eclipse.osgi\2416\0\.cp\lib\org.thingml.editor.standaloneApp-0.7.0-SNAPSHOT-jar-with-dependencies.jar
	at org.eclipse.osgi.framework.util.SecureAction.getZipFile(SecureAction.java:317)
	at org.eclipse.osgi.storage.bundlefile.ZipBundleFile.basicOpen(ZipBundleFile.java:93)
	at org.eclipse.osgi.storage.bundlefile.ZipBundleFile.getZipFile(ZipBundleFile.java:106)
	at org.eclipse.osgi.storage.bundlefile.ZipBundleFile.checkedOpen(ZipBundleFile.java:65)
	at org.eclipse.osgi.storage.bundlefile.ZipBundleFile.getEntry(ZipBundleFile.java:240)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathEntry.loadManifest(ClasspathEntry.java:142)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathEntry.<init>(ClasspathEntry.java:64)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.createClassPathEntry(ModuleClassLoader.java:257)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.createClassPathEntry(ClasspathManager.java:329)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.getClasspath(ClasspathManager.java:281)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.addStandardClassPathEntry(ClasspathManager.java:198)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.addClassPathEntry(ClasspathManager.java:190)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findClassPathEntry(ClasspathManager.java:173)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.buildClasspath(ClasspathManager.java:152)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.<init>(ClasspathManager.java:81)
	at org.eclipse.osgi.internal.loader.EquinoxClassLoader.<init>(EquinoxClassLoader.java:57)
	at org.eclipse.osgi.internal.loader.BundleLoader.createClassLoaderPrivledged(BundleLoader.java:293)
	at org.eclipse.osgi.internal.loader.BundleLoader.getModuleClassLoader(BundleLoader.java:231)
	at org.eclipse.osgi.internal.loader.BundleLoader.findEntries(BundleLoader.java:796)
	at org.eclipse.osgi.container.ModuleWiring.findEntries(ModuleWiring.java:289)
	at org.eclipse.osgi.storage.ManifestLocalization.findResource(ManifestLocalization.java:200)
	at org.eclipse.osgi.storage.ManifestLocalization.lookupResourceBundle(ManifestLocalization.java:131)
	at org.eclipse.osgi.storage.ManifestLocalization.getResourceBundle(ManifestLocalization.java:102)
	at org.eclipse.osgi.storage.ManifestLocalization.getHeaders(ManifestLocalization.java:63)
	at org.eclipse.osgi.storage.BundleInfo$Generation.getHeaders(BundleInfo.java:124)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.privGetHeaders(EquinoxBundle.java:462)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.getHeaders(EquinoxBundle.java:457)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.getHeaders(EquinoxBundle.java:451)
	at org.eclipse.ui.internal.about.AboutBundleData.getResourceString(AboutBundleData.java:86)
	at org.eclipse.ui.internal.about.AboutBundleData.<init>(AboutBundleData.java:41)
	at org.eclipse.ui.internal.about.ConfigurationLogDefaultSection.appendRegistry(ConfigurationLogDefaultSection.java:145)
	at org.eclipse.ui.internal.about.ConfigurationLogDefaultSection.write(ConfigurationLogDefaultSection.java:54)
	at org.eclipse.epp.internal.logging.aeri.ide.processors.ConfigurationInfo.getConfiguration(ConfigurationInfo.java:34)
	at org.eclipse.epp.internal.logging.aeri.ide.processors.PlatformDetailsSectionProcessor.canContribute(PlatformDetailsSectionProcessor.java:29)
	at org.eclipse.epp.internal.logging.aeri.ide.processors.CachingStringProcessor.canContribute(CachingStringProcessor.java:95)
	at org.eclipse.epp.internal.logging.aeri.ide.processors.ReportProcessorSafeWrapper.canContribute(ReportProcessorSafeWrapper.java:28)
	at org.eclipse.epp.internal.logging.aeri.ide.processors.Processors.canContribute(Processors.java:36)
	at org.eclipse.epp.internal.logging.aeri.ide.processors.Processors.shouldProcess(Processors.java:31)
	at org.eclipse.epp.internal.logging.aeri.ide.dialogs.ReviewDialog.lambda$3(ReviewDialog.java:282)
	at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:174)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1374)
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:481)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:471)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:151)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:174)
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:418)
[... 33 lines cut ...]
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:150)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:687)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:604)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:138)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
Caused by: java.util.zip.ZipException: error in opening zip file
	at java.util.zip.ZipFile.open(Native Method)
	at java.util.zip.ZipFile.<init>(ZipFile.java:219)
	at java.util.zip.ZipFile.<init>(ZipFile.java:149)
	at java.util.zip.ZipFile.<init>(ZipFile.java:163)
	at org.eclipse.osgi.framework.util.SecureAction.getZipFile(SecureAction.java:304)
	... 97 more

brice-morin, 2016-10-31:
OK, I pushed a new version, but did not have time to test it. Can you update in some 20 minutes to see if it helps?

brice-morin, 2016-10-31:
OK, I think I have identified the problem. We now use Git LFS for some files in the repository. Our build server does not have (yet) Git LFS installed. @ffleurey will update the server.

brice-morin, 2016-10-31:
Should be fixed now.

closed 2016-10-31 by brice-morin