TelluIoT/ThingML #134: Connector inconsistency to be detected

oysteinh, 2016-11-18:
**ThingML compiler suggestion**
Connector consistency does not seem to be checked.
Extracts of my example:
Definition of the connector in the configuration:
connector pim.request_sensor => T1.require_val
Then we have the appropriate definitions of ports:
required port request_sensor {
sends fetch_all_temps, fetch_temp, add_thermometer, fetch_all_hum, fetch_humidity, add_hygrometer}

and
provided port require_val {receives initialize,add_thermometer}

As we can clearly see the sending port can send messages that the receiving port cannot receive. Thus there is an inconsistency which should be pointed out by the compiler.
There is a potential consistency problem the other way as well. The receiving port can receive message “initialize”, but there are no sending end of any connector attached to request_sensor that sends that message. (This cannot be seen here with this extract, but will be known from the configuration file)

brice-morin, 2016-11-18:
You will now get messages like this: `[WARNING] ThingML: Message b is expected by instance a but never sent by instance b (in org.sintef.thingml.impl.ConnectorImpl@53348004 (name: null))`

or like that: `[WARNING] ThingML: Message a of instance b is sent but not received by instance a (in org.sintef.thingml.impl.ConnectorImpl@53348004 (name: null))` 

Note that it is not yet available in the Eclipse plugins

brice-morin, 2016-11-18:
Note that we just raise a Warning and do not block the compilation as sometimes it can be convenient to actually be able to only map a subset of a port (though we could extend the language to declare some messages as optional).

closed 2016-11-18 by brice-morin

brice-morin, 2016-12-05:
I updated the rule so that it considers the whole set of connectors connected to each instance rather than each connector individually, so as to avoid multiple warnings in the case of n-m connectors. Not available yet in the Eclipse plugins.

commit 68731026 (2016-11-18): fix #134