TelluIoT/ThingML #136: Allow the use of @c_header on Datatypes

jakhog, 2016-12-12:
When interfacing with external libraries, it seems useful to be able to wrap complex datatypes defined by the libraries with ThingML datatypes. However, this requires that library header files (at least in C/C++) are included wherever these types are used, which is currently not supported. A possible solution to this would be to allow the `@c_header` annotation on datatypes (and possibly objects) like the following:
```
datatype LEAdvData<32>
  @c_type "struct ble_adv_data_pack"
  @c_header "#include <bluetooth/bluetooth.h>";
```
and then include this header-line wherever the datatype is used.

I would also assume that a similar solution makes sense for other languages like Java with import.

brice-morin, 2016-12-12:
Yes, it makes sense for Java too. But if it is part of the standard library, putting the full qualified name in the `@java_type` annotation does the trick. If not, we have an annotation (a bit like `@c_header`) to include a Maven dependency. 

So I suggest you fix this issue yourself :-)

jakhog, 2016-12-14:
Fixed in b18797f4dadd7bd31aab5de104e18ad3560c694e

closed 2016-12-14 by jakhog