TelluIoT/ThingML #138: De-serialising bug in C compiler?

jakhog, 2017-01-05:
The following snippet of code is found in the `CCompilerContext.java` file [here](https://github.com/SINTEF-9012/ThingML/blob/1b16f90c9d23fd4b44858c41c983c8dbd440d27a/compilers/c/src/main/java/org/thingml/compilers/c/CCompilerContext.java#L710)
```
while (i > 0) {
    i = i - 1;
    if (i == 0) result += buffer + "[" + index + "]";
    else result += "(" + buffer + "[" + index + "]" + "<<" + (8 * i) + ") + ";
    index = index + 1;
}
```
@ffleurey @Lyadis; wouldn't this only work for integer types?

I would guess that the crazy union-trick should be used here as well?

Lyadis, 2017-01-05:
Are you sure this code is actually used? It looks like a previous (and not deleted) code for serialization.
But anyway, you are right, this should not work for double and float.

brice-morin, 2017-06-22:
@jakhog what is the status of this issue?

jakhog, 2017-06-22:
Fixed in 6c4e9c914b1470cc7757ec6e93307e1cfbbc33ba

closed 2017-06-22 by jakhog