TelluIoT/ThingML #141: Function call in property/variable initialization

ffleurey, 2017-01-17:
Seem to be be not understood properly by the checker (ClassCastException suspected)

brice-morin, 2017-01-17:
Actually, the checker is not responsible. The problem is in the C compiler, where `context.getConcreteThing()` returns `null`...

```java
public void generate(FunctionCallExpression expression, StringBuilder builder, Context ctx) {

        CCompilerContext context = (CCompilerContext) ctx;

        builder.append(context.getCName(expression.getFunction(), context.getConcreteThing()));

        builder.append("(_instance");
        for (Expression p : expression.getParameters()) {
            builder.append(", ");
            generate(p, builder, context);
        }
        builder.append(")");
    }
```

brice-morin, 2017-01-17:
The dirty hack is:
```java
        //FIXME: @ffleurey, @Lyadis this is a dirty hack... I get we should set the concreteThing in the context somewhere before...
        Thing concreteThing = context.getConcreteThing();
        if (concreteThing == null) {
            concreteThing = (Thing) expression.getFunction().eContainer();
        }
```

closed 2017-01-17 by brice-morin

brice-morin, 2017-01-17:
Seems like @Lyadis 's fix works 👍

commit 389112b0 (2017-01-17): fix #141 (with a dirty hack to be fixed by @ffleurey or @Lyadis)
commit 67b09801 (2017-01-17): Would that fix #141 in a preferable way than 389112b003616d21fe0468b769e9accf12714093?