TelluIoT/ThingML #150: Usage of threads in single-threaded POSIX compiler?

jakhog, 2017-01-25:
The single-threaded POSIX compiler generates code for thread-safe access to the fifo. This is a little bit confusing to me. Is the idea that threads should still be used in the POSIX compiler (e.g. in network plugins), but that all the state machines live in a single thread?

Somewhat related; the way that processMessageQueue() is implemented now breaks polling code. The function will block until a message is put on the fifo.

jakhog, 2017-01-26:
So I guess the answer to my question is yes.

And for the second part, no polling code should be added to the POSIX compiler, and the current behavior is OK. Since any events has to come through the fifo.

closed 2017-01-26 by jakhog

Lyadis, 2017-01-26:
Short answer: yes network listener can be in separated thread in the posix compiler, and they do post messages in the fifo.

Long answer:
I think it is just a matter of legacy architecture. The arduino compiler produced full single threaded code (no other choice), and posix had network listener that could post messages from separated thread. But as most code was shared by both compilers, all instances remained in the same thread.
Then we added the multi thread posix compiler that create one thread per instance. I think the good question concerning the posix (pseudo) single thread compiler, is what target does it aim, and should it be fully single threaded (even for network listener).