TelluIoT/ThingML #152: Common namespace for message names and function names

sdalgard, 2017-02-02:
When generating code for javascript there is name collision for messages and functions
No error messages, but the generated code is not executable.

Suggest to generate code where function names are unique by adding a prefix.

```
internal event m : BLEconnect?doDisconnect
action do	        	
    doDisconnect(m.handle)
end

```

brice-morin, 2017-02-02:
Yes. I think I will generate function called `f_myFunction`

brice-morin, 2017-02-02:
Thinking a bit more about it and trying to reproduce the problem, I do not see how a naming conflict could happen between a function and a message having the same name..... I suspect your compiler is not up-to-date, as I recently made changes around functions, so that a function call is now `this.myFunction()` in the generated code rather than `myFunction()` (which could indeed colllide with a message of the same name).

closed 2017-02-02 by brice-morin

sdalgard, 2017-02-02:
Hi here is a snippet of the generated js code
I have changed the function name to be able to run the code

```
internal event m : BLEconnect?doDisconnect
action do	        	
    doDisconnectF(m.handle)
end

```
```
BLE_BLEImpl_Scanning.to(null).when(function (doDisconnect) {return doDisconnect._port === "BLEconnect" && doDisconnect._msg === "doDisconnect";}.bind(this)).effect(function (doDisconnect) {
doDisconnectF.call(this, doDisconnect.handle);
}.bind(this));
```

It is the .call that fails

sdalgard, 2017-02-02:
Hi I am not sure if we are observing the same generated snippet.
I added this to the closed case...
Hi here is a snippet of the generated js code
I have changed the function name to be able to run the code
internal event m : BLEconnect?doDisconnect
action do
    doDisconnectF(m.handle)
end

BLE_BLEImpl_Scanning.to(null).when(function (doDisconnect) {return doDisconnect._port === "BLEconnect" && doDisconnect._msg === "doDisconnect";}.bind(this)).effect(function (doDisconnect) {
doDisconnectF.call(this, doDisconnect.handle);
}.bind(this));
It is the .call that fails

brice-morin, 2017-02-02:
Yes, I agree it *was* a problem, but the changes recently made to the JS compiler have made this conflict impossible. Try updating your Eclipse.