TelluIoT/ThingML #154: Cannot read property 'length' of undefined

sdalgard, 2017-02-06:
Problem when running generated js code

```
/root/homemnt/heads/thingml-gen/nodejs/FieldNode/lib/SensorTimerJS.js:60
const arrayLength = self.timer_timeoutOntimerListeners.length;
                                                      ^

TypeError: Cannot read property 'length' of undefined
    at sendTimer_timeoutOnTimer (/root/homemnt/heads/thingml-gen/nodejs/FieldNode/lib/SensorTimerJS.js:60:55)
    at processImmediate [as _immediateCallback] (timers.js:383:17)
root@ubilinux:~/homemnt/heads/thingml-gen/nodejs/FieldNode/lib# 

```

[sensortimer.zip](https://github.com/SINTEF-9012/ThingML/files/753941/sensortimer.zip)

brice-morin, 2017-02-06:
Is your compiler up to date? We do not use these `self.<msg>On<Port>Listeners` anymore... I did a big refactoring last week, to remove all the `.bind` in the genrated code and I also changed the internal communication (listeners) + some other things to try to make the integration with JS code simpler.

brice-morin, 2017-02-06:
Also, I have been using this exact SensorTimerJS component with the new compiler and it is working. I probably changed something in the way we mix JS and ThingML, so here is the source:

```
import "../SensorTimer.thingml"

// Manage a set of software timers.
thing SensorTimerJS includes SensorTimer
@pim "SensorTimer"
@platform "javascript"
@javascript "node browser"
{
    function cancel() 
    do
        'clearTimeout(this.timer);'
    end
    
    function starting(delay : Integer)
    do
        'this.timer = setTimeout(()=>{' 
        timer!timer_timeout() 
        '},' & delay & ');'
    end
    
    function startInterval(interval : Integer)
    do
        'this.timer = setInterval(()=>{'
        timer!timer_timeout() 
        '},' & interval & ');'
    end
    
    /*function onTimeout() 
    do
        timer!timer_timeout()
    end*/
    
    statechart SoftTimer init default {
        state default {
            
          internal startState
            event m : timer?timer_start
            guard m.delay > 0
            action do
                starting(m.delay)
            end
            
           internal startInterval
            event m : timer?timer_start_interval
            guard m.interval > 0
            action do
                startInterval(m.interval)
            end
            
          internal cancel
            event m : timer?timer_cancel
            action cancel()    
        }
    }
}

```

sdalgard, 2017-02-06:
I am running 
ThingML 0.7.0
@brice-morin brice-morin released this on 4 Jan · 126 commits to master since this release

ThingML 0.7.0, Copyright SINTEF, released under Apache 2.0 license
Downloads
5.92 MBorg.thingml.eclipse.updatesite-0.7.0.zip

That is the latest zip at https://github.com/SINTEF-9012/ThingML/releases.

Eclipse also report 0.7.0 after check for updates.

Where can I find the new release?

brice-morin, 2017-02-06:
You have to use the update site that contains the latest version of ThingML, not the release, as detailed at the end of the [Installation section](https://github.com/SINTEF-9012/ThingML#-installation)

closed 2017-02-06 by brice-morin