TelluIoT/ThingML #156: CharLiteral, ByteLiteral

brice-morin, 2017-02-08:
We need to be able to express characters and bytes directly in ThingML. Right now, we use extern expression to express a byte or character literal, which makes a lot of things to appear as PSM whereas they are PIM (we can qualify a thing as PIM if does not contain any extern expression of extern statement).

One problem is that characters are typically expressed as `'c'` in most (if not all) programming languages. We use single quotes for externs... We might want to use backquotes for externs (e.g. as code snippet in MarkDown) and stick with a conventional notation for characters.

jakhog, 2017-02-08:
I agree that we should support character literals, and also that - if we do - they should really be the same as most other programming languages (single quotes). Changing the externs to backquotes breaks a lot of old ThingML code, but at the same time, we wouldn't have to escape all single quotes in externs, which is a nice side-effect.

But what would a byte literal be @brice-morin? Wouldn't that just be a number? Or are you thinking of supporting hex notations like `0xFF` - which in my mind should just be the same value as `255`.

Isn't this also somewhat related to having a set of basic datatypes included in the language?

brice-morin, 2017-02-08:
Yes, I was thinking about `0x..` notation (which you use, with quotes e.g. in your ThingML-BLE project), but we could indeed fallback to numbers. Not sure e.g. in Java if we can assign 255 to a byte directly or whether we should use the hex notation (though the compiler could take care of the conversion)...

At least we should have proper literals for all common primitive types, which well accepted notations.

Since we agree, @ffleurey has to fix it :-)

brice-morin, 2017-02-17:
Also, 'BooleanLiteral' should have a 'boolValue' of type 'EBoolean' and not 'EString' (not that it is a big deal to convert between the two, that is just annoying and inconsistent with other Literals)

[image]

jakhog, 2018-01-23:
@ffleurey, is there a good reason no to remove single quotes (') from externs and use them for character literals right now? (now that we can use backticks (`) for externs?

It will break some old code, but as far as I have understood, this is what we want to do at some point either way, and it would be quite an easy fix for old code...

brice-morin, 2018-01-25:
I guess this is fixed?

closed 2018-01-25 by brice-morin