TelluIoT/ThingML #158: Region.getSubstate() ???

brice-morin, 2017-02-13:
It seems `Region.getSubstate()` does not exist anymore in the new metamodel (XText). Which class now contains sub-state?

It seems that `Region.getSubstate()` is one of the last blocking issue before we can re-try the Java and JavaScript compilers...

jakhog, 2017-02-13:
It seems like it was moved (back) to the `CompositeState` class...

brice-morin, 2017-02-13:
Yes. But what is a region now in the metamodel. Seems like we can just get and set an initial state, but no more. 

Also, sessions used to be states. Not anymore.

I guess we have to think and decide a few things on this part of the metamodel (the rest is somehow similar modulo minor adjustments) before we can finalize the migration of compilers...

brice-morin, 2017-02-17:
OK, it pretty much migrated everything I could in all compilers and plugins. Still, we have a problem with Composite, Region and Sessions. I made a picture suggesting refactorings.

[image]

ffleurey, 2017-02-21:
Solved by a complete refactoring of the metamodel around states, sessions and regions.

closed 2017-02-21 by ffleurey