TelluIoT/ThingML #160: List of features

brice-morin, 2017-02-17:
- [ ] Abstract functions so as we can have PIM statecharts and just implement functions in a PSM thing
- [ ] external things to replace protocols. Things can define ports and properties and be instantiated many times
- [ ] Byte/Bit operations
- [ ] composite types (struct like) packed (array) and not packed (memory-aligned)
- [ ] Prevent a thing from including two statecharts
- [ ] for each loop for arrays (maybe two types: to iterate over elements and to iterate over indexes)
- [ ] dafault/basics datatypes 
- [ ] Time/Timers directly in the language
- [ ] CEP

jakhog, 2017-02-17:
### Some thoughts about for each loops
It would be nice to have two different syntaxes if you want to loop over the variable or the indices, like:
```
foreach ( value in array ) do
   ...
end
```
(the type of the value is given by the type of the array) and:
```
foreach ( index i : Integer in array ) do
   ...
end
```

And maybe break/continue in both foreach and while loops :)

skorsky, 2017-02-20:
CEP as a feature: does this mean based on ReactiveX or in connection with Apama? What are the plans for a feature?

ffleurey, 2017-02-21:
This is more a a meta-issue and whould be broken down in individual feature requests