TelluIoT/ThingML #161: Internal Port

brice-morin, 2017-02-20:
Internal port allows defining ports internal to a thing, that are connected to themselves by default (no need to do it in the configuration). Basically, it allows internal async communication. Right now, internal ports have a similar syntax as other port, with a list of sent and received messages. The thing is, internal ports typically (and should always) be able to receive what they send (i.e. both lists are equal). We should simplify the syntax (and maybe adapt the metamodel) to only declare one list.

ffleurey, 2017-02-21:
This is on hold until better specified

brice-morin, 2017-02-21:
I suggest something like:

```
internal port p {
  handles m1, m2
}
```

Ideally (not to change all the compilers), the list of handled messages should be copied into the sent AND the received messages somewhere in the parser

jakhog, 2017-02-21:
Why not `messages` instead of `handles`?

```
internal port p {
  messages m1, m2
}
```

I guess if we want something else on the ports `handles` is better, but for now `messages` is quite clear.

brice-morin, 2017-02-21:
That could work too