TelluIoT/ThingML #162: Outline is nearly useless

brice-morin, 2017-02-20:
I am not too impressed by the new outline... Either the parser is not working properly, or the outline is not complete, but it seems my models are lacking quite a few elements in the outline, and I cannot see anything in the properties view. Quite annoying, especially as we are migrating to the new parser...

ffleurey, 2017-02-21:
The EMF "sample reflexive editor" can be used to inspect the models. The outline will be improved later but has no ambition to exhibit the whole model

brice-morin, 2017-08-04:
@jakhog is working on it and it looks promising :-)

brice-morin, 2017-08-09:
I did some more work on the outline, resulting in two conflicting visual styles. I suggest we use the black&white icons. But instead of the black, we might use the color of the thingml keyword, so that we can more easily map the concepts between the outline and the model.

brice-morin, 2018-01-09:
Not 100% perfect, yet. But still acceptable. Closing the issue.

closed 2018-01-09 by brice-morin