TelluIoT/ThingML #175: Saving/serializing ThingML models with XText

jakhog, 2017-04-27:
After loading a ThingMLModel, saving it to a file (either ThingML or XMI) using `ThingMLCompiler.saveAs...(model, file)`, I get the following error and output:
```
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
No valid model found for ressource: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testJar/src/main/resources/tests/datatypes.thingml
Unable to load resource ../thingml.thingml
Exception in thread "main" java.lang.RuntimeException: No EObjectDescription could be found in Scope Thing.includes for ThingMLModel.types[2]->Thing'Test'
Semantic Object: ThingMLModel.types[0]->Thing'TestInt16'
URI: file:/C:/Users/jakobho/Documents/GitHub/ThingML-xtext/testing/tmp/genCode/Types/Int16.thingml
EStructuralFeature: thingML::Thing.includes
        at org.eclipse.xtext.serializer.diagnostic.ISerializationDiagnostic$ExceptionThrowingAcceptor.accept(ISerializationDiagnostic.java:131)
        at org.eclipse.xtext.serializer.tokens.CrossReferenceSerializer.getCrossReferenceNameFromScope(CrossReferenceSerializer.java:138)
        at org.eclipse.xtext.serializer.tokens.CrossReferenceSerializer.serializeCrossRef(CrossReferenceSerializer.java:111)
        at org.eclipse.xtext.serializer.acceptor.SequenceFeeder.getToken(SequenceFeeder.java:481)
        at org.eclipse.xtext.serializer.acceptor.SequenceFeeder.accept(SequenceFeeder.java:262)
        at org.eclipse.xtext.serializer.sequencer.BacktrackingSemanticSequencer.accept(BacktrackingSemanticSequencer.java:434)
        at org.eclipse.xtext.serializer.sequencer.BacktrackingSemanticSequencer.createSequence(BacktrackingSemanticSequencer.java:501)
        at org.thingml.xtext.serializer.ThingMLSemanticSequencer.sequence_Thing(ThingMLSemanticSequencer.java:2185)
        at org.thingml.xtext.serializer.ThingMLSemanticSequencer.sequence(ThingMLSemanticSequencer.java:346)
        at org.eclipse.xtext.serializer.sequencer.AbstractSemanticSequencer.createSequence(AbstractSemanticSequencer.java:67)
        at org.eclipse.xtext.serializer.acceptor.SequenceFeeder.acceptEObjectRuleCall(SequenceFeeder.java:325)
        at org.eclipse.xtext.serializer.acceptor.SequenceFeeder.acceptRuleCall(SequenceFeeder.java:352)
        at org.eclipse.xtext.serializer.acceptor.SequenceFeeder.accept(SequenceFeeder.java:263)
        at org.eclipse.xtext.serializer.sequencer.BacktrackingSemanticSequencer.accept(BacktrackingSemanticSequencer.java:434)
        at org.eclipse.xtext.serializer.sequencer.BacktrackingSemanticSequencer.createSequence(BacktrackingSemanticSequencer.java:501)
        at org.thingml.xtext.serializer.ThingMLSemanticSequencer.sequence_ThingMLModel(ThingMLSemanticSequencer.java:2157)
        at org.thingml.xtext.serializer.ThingMLSemanticSequencer.sequence(ThingMLSemanticSequencer.java:349)
        at org.eclipse.xtext.serializer.sequencer.AbstractSemanticSequencer.createSequence(AbstractSemanticSequencer.java:67)
        at org.eclipse.xtext.serializer.impl.Serializer.serialize(Serializer.java:115)
        at org.eclipse.xtext.serializer.impl.Serializer.serialize(Serializer.java:128)
        at org.eclipse.xtext.serializer.impl.Serializer.serialize(Serializer.java:192)
        at org.eclipse.xtext.resource.XtextResource.doSave(XtextResource.java:386)
        at org.eclipse.emf.ecore.resource.impl.ResourceImpl.save(ResourceImpl.java:1430)
        at org.eclipse.emf.ecore.resource.impl.ResourceImpl.save(ResourceImpl.java:999)
        at org.thingml.compilers.ThingMLCompiler.save(ThingMLCompiler.java:170)
        at org.thingml.compilers.ThingMLCompiler.saveAsXMI(ThingMLCompiler.java:178)
        at org.thingml.testing.commandline.Main.main(Main.java:198)
```

brice-morin, 2017-04-27:
Seems like dependent models (e.g. datatypes.thingml) using import and not properly put in the Resource Set when saving, breaking some references (e.g. property cannot resolve their types). We should make sure we add all dependent resource when saving.

jakhog, 2017-04-27:
I don't know if it's relevant, but I don't want to save the `import "..."` statement. I would like it to save the full ThingML model so that it can be loaded again from this single file.

Is the `import "..."` statement part of the ThingML model, or is it simply a keyword for the parser to also include other files while parsing the file?

brice-morin, 2017-06-28:
I guess this is solved now? If not, re-open

closed 2017-06-28 by brice-morin

jakhog, 2017-06-28:
Well, yes, kind-of. I was trying to save .thingml-files with imports, which you can now do with flattening.

But saving the original models still doesn't work. But I'm completely fine with that :metal: