TelluIoT/ThingML #176: How to include dependency in Browser JS?

brice-morin, 2017-04-27:
I want to use [this lib](https://github.com/Tonejs/Tone.js/), also available on [NPM](https://www.npmjs.com/package/tone), in a ThingML program (I guess you know which one...).

What is the best way to rapidly include a third-party dependency in a ThingML program targeting the Browser. While the Node.JS compiler uses by default a proper and very common NPM packaging to deal with deps, it seems the Browser compiler just got rid of that... But I do not really want to always and manually copy/paste my deps when I generate code...

So?

jakhog, 2017-04-27:
Well, that's something we should make a good permanent solution for... I remember that we had an issue somewhere where we got quite a lot of comments about this, but I can't seem to find it right now...

It kind of depends on what we would like the target to be for the browser compiler. If we want to use it offline, we need some packaging stuff. If we want to target mostly online, I would like to simply allow including scripts from something like `cdnjs.com` or similar, just like you would do for any other webapp. E.g. in your case `<script src="https://cdnjs.cloudflare.com/ajax/libs/tone/0.10.0/Tone.js"></script>`.

Then we don't need any extra steps.

At least we can start with that, and see if we need a proper packaging system later?

brice-morin, 2017-04-27:
Seems OK. Meanwhile I implemented a dirty workaround :-) But it should be very easy then to switch to cdnjs-based URLs

jakhog, 2017-04-28:
I'm changing it a bit right now, but you are actually copying the files to the local directory. Is that what we want?

Is there a way to pass arbitrary options to the compiler when running it? This specific example seems like an example where you would like to decide if you copy the files locally when you actually run the compiler (not in an annotation).

brice-morin, 2017-04-28:
No, we do not really want to copy the files to the local directory, it was just a workaround. We could use a `@cdnjs` annotation, with just the URL, in order to populate the scripts in index.html, as you suggested. Then we do not need to copy anything.

Or we can have both options.

Note that compilers implement this method: `public abstract void compile(Configuration cfg, String... options);` so yes, it should be possible to pass extra options to compilers. Not sure how it works from the CLI

jakhog, 2017-04-28:
OK, then ill do the simple include thing (it should be just an url I think, not just cdnjs).

All have a look at the options as well, maybe it's usefull somewhere else...

brice-morin, 2017-04-28:
Yes, makes sense to generalize to `@url`

brice-morin, 2017-04-28:
Also I was thinking, it might be an idea to put this annotation on `object` in ThingML as well. Typically we use `object` as a pointer to an external lib, rather than simply putting this annotation on the thing itself, which can include several libs (though we can have multiple annotations of the same type)

jakhog, 2017-04-28:
Good point!

commit 8503d873 (2017-04-28): Added @js_include annotation to include external JS libraries. Closes #176

It is now possible to add @js_include annotation on either a
Configuration, a Thing, or an Object in ThingML.

The syntax is one of:
  @js_include "<src>"
  @js_include "src=<src>,target=<target>"
  @js_include "target=<target>,src=<src>"
Which compiles to:
  <script type="application/javascript" src="<src>"[ target="<target>"]></script>
in the index.html file.
closed 2017-04-28 by jakhog