TelluIoT/ThingML #177: Not retrieving all types in some helpers?

jakhog, 2017-04-28:
While going through some helper functions, it seems that we might not be retrieving all types in some of them. E.g. in [allUsedTypes](https://github.com/SINTEF-9012/ThingML/blob/8503d8738409a13745a55e84ed1dbc3c1c359de7/language/thingml/src/org/thingml/xtext/constraints/ThingMLHelpers.java#L254) here:
```
for(Property p : ThingHelper.allPropertiesInDepth(thing)) {
    if (EcoreUtil.equals(p.getTypeRef().getType(), t))
        result.add(t);
    }
    for(Message m : ThingMLHelpers.allMessages(thing)) {
        for(Parameter p : m.getParameters()) {
            if (EcoreUtil.equals(p.getTypeRef().getType(), t)) {
                result.add(t);
            }
        }
    }
}
```

But then we are missing the types for the variables in actions and functions right?

Also might be a problem in [allUsedSimpleTypes](https://github.com/SINTEF-9012/ThingML/blob/8503d8738409a13745a55e84ed1dbc3c1c359de7/language/thingml/src/org/thingml/xtext/constraints/ThingMLHelpers.java#L275), [allObjectTypes](https://github.com/SINTEF-9012/ThingML/blob/8503d8738409a13745a55e84ed1dbc3c1c359de7/language/thingml/src/org/thingml/xtext/helpers/ConfigurationHelper.java#L275).

brice-morin, 2017-04-28:
Yes, I guess `allPropertiesInDepth` is limited to property/var declared in things and states, but does not go deeper. We should also look at functions params, like we do for messages, but also in the detailed code for  local vars, etc, as you mention. We have some utilities to find all expressions or all actions of a given type, which can help e.g. `ThingMLHelpers.getAllExpressions(t, PropertyReference.class)` to get all PropertyReference anywhere in thing t (it will go recursively down to eveything that is contained directly or indirectly by t)

brice-morin, 2017-04-28:
e.g. you should look for all actions of type LocalVariable in addition to the rest

commit 13729591 (2017-05-09): fix #177
closed 2017-05-09 by brice-morin