TelluIoT/ThingML #180: Could I display the transitions names in the PlantUML state-machine chart ?

yellowiscool, 2017-05-15:
## What I have

[image]

## What I want

[image]

## My ThingML code

```thingml
       statechart ChatBot init start_conversation {
             
             state start_conversation {
                    transition -> start_new_topic
             }
             
             state start_new_topic{
                    transition should_start_small_talk -> small_talk
                    transition should_ask_for_category -> ask_for_category
                    transition user_wants_to_leave_conversation -> goodbye
             }
             
             state small_talk{
                    transition should_continue_small_talk -> small_talk
                    transition should_ask_for_category -> ask_for_category
                    transition should_continue_ingoing_conversation -> talk_about_topic
             }
             
             state ask_for_category{
                    transition vague_category -> start_new_topic
                    transition has_identified_category_correctly -> talk_about_topic
             }
             
             state talk_about_topic{
                    transition should_continue_conversation -> talk_about_topic
                    transition shouldnot_continue_conversation -> start_new_topic
                    transition user_wants_small_talk -> small_talk
                    transition user_wants_to_leave_conversation -> goodbye
             }
             
             state goodbye{}     
       }  
```

## The rabbit

🐇

brice-morin, 2017-05-15:
Yes, should be quite easy...

brice-morin, 2017-05-15:
which version of ThingML are you using? The one based on EMFText, or the one based on XText (ThingML v2)?

yellowiscool, 2017-05-15:
EMFText because I followed the documentation, but I will compile the XText version very soon. So assume it's XText.

brice-morin, 2017-05-15:
OK, should be fixed in XText branch

yellowiscool, 2017-05-15:
Some wild `\n` appeared.

```
@startuml
skinparam defaultTextAlignment left
caption Behavior of thing ChatBotSM
[*] --> ChatBot
state ChatBot{
state start_conversation{

start_conversation --> start_new_topic
}
state start_new_topic{

start_new_topic --> small_talkshould_start_small_talk\n

start_new_topic --> ask_for_categoryshould_ask_for_category\n

start_new_topic --> goodbyeuser_wants_to_leave_conversation\n
}
state small_talk{

small_talk --> small_talkshould_continue_small_talk\n

small_talk --> ask_for_categoryshould_ask_for_category\n

small_talk --> talk_about_topicshould_continue_ingoing_conversation\n
}
state ask_for_category{

ask_for_category --> start_new_topicvague_category\n

ask_for_category --> talk_about_topichas_identified_category_correctly
}
state talk_about_topic{

talk_about_topic --> talk_about_topicshould_continue_conversation\n

talk_about_topic --> start_new_topicshouldnot_continue_conversation\n

talk_about_topic --> small_talkuser_wants_small_talk\n

talk_about_topic --> goodbyeuser_wants_to_leave_conversation\n
}
state goodbye{
}
[*] --> start_conversation
}
@enduml
```

But when I removed them, I got something strange. I guess something is messing between the next state and the transition's name.

[image]

brice-morin, 2017-05-15:
Hmmmm, those \n, \\n, etc is quite a pain in plantUML :-) But I will try to fix that

brice-morin, 2017-05-15:
this time, it should work :-)

commit b4ce75a1 (2017-05-15): fixes #180
closed 2017-05-15 by brice-morin