TelluIoT/ThingML #181: Updatesite doesn't build on the xtext HEAD

yellowiscool, 2017-05-15:
Maven fails with the following error : 
```
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.thingml.eclipse.updatesite raw:1.0.0.'SNAPSHOT'/format(n[.n=0;[.n=0;[-S]]]):1.0.0-SNAPSHOT
[ERROR]   Missing requirement: org.thingml.eclipse.updatesite raw:1.0.0.'SNAPSHOT'/format(n[.n=0;[.n=0;[-S]]]):1.0.0-SNAPSHOT requires 'org.thingml.eclipse.feature.feature.group [1.0.0,1.0.1)' but it could not be found
```

yellowiscool, 2017-05-15:
Let's never talk about this anymore.

yellowiscool, 2017-05-15:
Thanks @ffleurey btw.

brice-morin, 2017-05-15:
Was the issue between your screen and your chair? :-)

yellowiscool, 2017-05-15:
I stupidly followed the instructions from the readme. The instructions were wrong, but I should have seen it.

jakhog, 2017-05-16:
As far as I understood, the instructions in the README.md is for the EMF version, so it should really be fixed on the xtext branch?

brice-morin, 2017-05-16:
README.md in xtext branch definitely needs some updates. Feel free to contribute :-)

commit 7c3d5480 (2018-01-11): Update README.md

Fixes #181
closed 2018-01-11 by jakhog