TelluIoT/ThingML #183: Running generated code on Arduino

imberium, 2017-06-16:
Hello
I generated code from the LED.thingml example and tried to run it on an Arduino MEGA ADK then I got this errors :

```
wiring.c.o (symbol from plugin): In function '__vector_23':

(.text+0x0): multiple definition of '__vector_23'

sketch\ArduinoSketchCfg.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status
```

Can you see what is the problem here ?

Thanks in advance.

brice-morin, 2017-06-19:
I suspect you are running some old samples that do not work anymore... @jakhog @ffleurey can you provide a simple blink example that works?

ffleurey, 2017-06-19:
Hello, indeed these examples are are probably outdated. I have committed some better ones here: https://github.com/ffleurey/ThingMLArduinoDemo

Have a look at those and let us know if you have any issues.

brice-morin, 2017-06-21:
I assume it works with the new examples. Just re-open this issue if not.

closed 2017-06-21 by brice-morin

imberium, 2017-06-23:
It works ! Thanks