TelluIoT/ThingML #186: Issue backslash not removed from annotation

madkira, 2017-06-26:
The generator leaves backslashes \ on annotation

```
@c_header "
#define SIMPLE_CLICK \"Simple Click\""
```
should generate 
```
#define SIMPLE_CLICK "Simple Clic"
```

but the result is  
```
#define SIMPLE_CLICK \"Simple Click\"
```

jakhog, 2017-06-27:
I checked, and the same happens for all string literals. E.g. if I make a property
```
property test : String = "\"Some string\""
```
It will generate this C-code:
```c++
t_var.Test_test_var = "\"Some string\"";
```
But in that case it actually makes sense, since we are adding the outer "" back.

So, maybe the actual fix would be to treat annotation values different from string literals? In a way, it isn't really a string literal...

jakhog, 2017-06-27:
Or, another thought:
We could (should?) unescape the string literals when we parse it (since the backslashes are not really a part of the actual string we want), and escape it back when we generate the code.

This might be even more correct, if you consider languages that doesn't use "" to express string literals.
You could choose to use single quotes in e.g. JavaScript...

brice-morin, 2017-06-27:
I know it has been some back and forth changes in the way we by default escape quotes in annotations, extern expressions and maybe string literal. I guess we should indeed treat string literals different from annotations. @ffleurey what is now the (un)escaping strategy in:

- String literal,
- annotation, and
- extern?

commit 31b393ce (2017-12-28): Added new back-tick terminal in the grammar

This type of string `...` can be used in both annotations and external
statements and expressions, should fix some issues with escaping in
annotations, and make it easier to use single quotes in externals.

The back-ticks are the same syntax as Markdown uses for code snippets,
which should be familiar to lot's of people.

It is also a step towards using '.' for characters in ThingML.

Fixes #186
Related to #156
closed 2017-12-28 by jakhog