TelluIoT/ThingML #187: Annotations doc

imberium, 2017-06-29:
Hello

Is there any documentation for the annotations that you are using ?

(I could find this https://github.com/TelluIoT/ThingML/blob/master/Documentrospection/Annotation.html but it is incomplete)

Thanks

jakhog, 2017-06-29:
Unfortunately, they are not very well documented, as they are evaluated at run-time when you compile your ThingML model to a target.

As you have already seen, there is the Documentrospection code that reads all the Java-source and tries to find places where we use specific annotations. But as you have noticed, it was last updated in October 2016, so there might be some changes.

If you are compiling ThingML from source, you can run the Documentrospection yourself to generate it again from the current Java source code.

Is there any specific ones that you are missing?

imberium, 2017-06-29:
Actually several ones, I thought that it could be better to have this doc especially when it comes to the platform specific ones as usually people will not be experts of the platform. It may avoid to dive into the code trying to figure it out

closed 2020-06-02 by brice-morin