TelluIoT/ThingML #192: Something missing in the sources ?

imberium, 2017-09-26:
Hello

I compiled ThingML from sources. The Jar produced here : /compilers/registry/target doesn't load any network plugin nor serialization ones and doesn't take into account CEP, it doesn't recognise the keyword "stream".

I also noticed that the syntax for CEP is not present in the Xtext grammar : /language/thingml/src/org/thingml/xtext/ThingML.xtext

But the ThingML plugin that you provide, works fine in Eclipse, it does actually take into account streams and loads all the network plugins.

Isn't there something missing in the sources ?

imberium, 2017-09-26:
This what I did, but still it doesn't take into account the CEP. As I said there no mention to the CEP syntax in : /language/thingml/src/org/thingml/xtext/ThingML.xtext

And also the eclipse plugin generated from the sources seem to use the one in : /compilers/registry/target.

brice-morin, 2017-09-26:
CEP is not supported in the current version of ThingML

imberium, 2017-09-26:
Ok Thanks for you reply.

What about the Eclipse plugin generated from sources, it doesn't load any plugin too ?

ffleurey, 2017-09-26:
If you follow the instructions to compile ThingML from sources (run maven in root directory and then run maven in language directory), you will find the compiled eclipse plugins in ThingML/language/thingml.repository/target. It is the complete update site, you can install the plugins direcly in eclipse and the plugins should be there.

imberium, 2017-09-27:
When I use ThingML v1, plugins are loaded correctly and the code generation is okay too, while when I use v2, plugins are not loaded and there are errors during the code generation.

I am using the packages respectivly from : http://thingml.org/dist/update/ and http://thingml.org/dist/update2/

jakhog, 2017-09-27:
That is strange, it works for me...

What version of Eclipse are you using? And what is the error you get during code generation?

imberium, 2017-09-27:
I got this error
```
Please contact the ThingML development team (though GitHub's issue tracker) with 1) your input model, and 2) the following stack trace:
FATAL ERROR: Exeption calling ThingML Compiler: org.apache.commons.io.IOUtils.toString(Ljava/io/InputStream;Ljava/nio/charset/Charset;)Ljava/lang/String;
	org.thingml.compilers.Context.getTemplateByID(Context.java:268)
	org.thingml.compilers.c.CCompilerContext.getThingHeaderTemplate(CCompilerContext.java:251)
	org.thingml.compilers.c.CThingApiCompiler.generateCHeader(CThingApiCompiler.java:97)
	org.thingml.compilers.c.CThingApiCompiler.generatePublicAPI(CThingApiCompiler.java:83)
	org.thingml.compilers.c.CExternalThingEnabledApiCompiler.generatePublicAPI(CExternalThingEnabledApiCompiler.java:44)
	org.thingml.compilers.c.posix.PosixCompiler.do_call_compiler(PosixCompiler.java:79)
	org.thingml.compilers.utils.OpaqueThingMLCompiler.compile(OpaqueThingMLCompiler.java:75)
	org.thingml.compilers.ThingMLCompiler.compile(ThingMLCompiler.java:294)
	org.thingml.eclipse.ui.commands.CompileThingFile.execute(CompileThingFile.java:261)
	org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:291)
	org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:92)
	sun.reflect.GeneratedMethodAccessor86.invoke(Unknown Source)
	sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	java.lang.reflect.Method.invoke(Unknown Source)
	org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:305)
	org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:239)
	org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:132)
	org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:152)
	org.eclipse.core.commands.Command.executeWithChecks(Command.java:494)
	org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:210)
	org.eclipse.ui.internal.handlers.LegacyHandlerService.executeCommand(LegacyHandlerService.java:390)
	org.eclipse.ui.menus.CommandContributionItem.handleWidgetSelection(CommandContributionItem.java:820)
	org.eclipse.ui.menus.CommandContributionItem.lambda$3(CommandContributionItem.java:797)
	org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:86)
	org.eclipse.swt.widgets.Display.sendEvent(Display.java:4428)
	org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1079)
	org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4238)
	org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3817)
	org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1155)
	org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1044)
	org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:680)
	org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:594)
	org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:151)
	org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	java.lang.reflect.Method.invoke(Unknown Source)
	org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
	org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
	org.eclipse.equinox.launcher.Main.run(Main.java:1499)

```

jakhog, 2017-09-27:
Well, that doesn't really explain too much about what the problem is...

Can you post the ThingML source that you're trying to compile?
Also, what Eclipse version are you using?

imberium, 2017-09-28:
I tried it on Eclipse Neon and Eclipse Oxygen and got the same error.
I got this error with this simple code : 
```
thing Hello {
	statechart myStatechart init one {
		state one {
			on entry do print "Hello world" end 
		}
	}

}
configuration helloexec 
@arduino_stdout "Serial"
{
	instance hello:Hello
}
```

imberium, 2017-09-29:
I ended up using your docker container on Ubuntu and it worked well.

But still I don't see what was the issue with the default setup. Maybe because I was running it on Windows.

Thanks for your help 👍

jakhog, 2017-09-29:
It seems like something specific to your setup, because neither of us were able to reproduce the error with our setups here... Very strange.

Glad you got it working!

imberium, 2017-09-29:
Do you have a mailing list or a working group, so that we can contribute efficiently ?