TelluIoT/ThingML #194: Checker performances issues in eclipse

ffleurey, 2017-10-05:
The checker seem to trigger way too often (or take too much time). Working on this project https://github.com/ffleurey/ThingML-PongTutorial in eclipse is almost impossible because for every time on file is checked, all the thingml files in the workspace are re-parsed and checked.
Before the checker was improved to catch more errors, this was not much of a problem. We should: 

1/ Make sure that we do not run the checker more than needed (it may be that xtext default behavior is to re-pared and validate too often) and 

2/ we should use a profiler on the checker to make sure that there are no big inefficiencies in it.

brice-morin, 2017-10-05:
Yes, I think I have experienced the same issues on the same project :-) I guess the default strategies should be to check on save, or something like this.

Well, the checker is deeply inefficient. Each rule basically re-browse the whole configuration and looks for stuff it is interested in. I think a better approach would be:
- only browse the configuration (and everything it contains or refers to) once
- each rule subscribes to model elements of interests (e.g. functions)
- the main "browser" notifies the rules with only the elements they are interested in
- each rule can just check what it is supposed to check, without the need for it to first find the elements of interests.

brice-morin, 2017-10-05:
Feel free to fix it, as I cannot allocate time on ThingML anymore... Or maybe I can have a look this weekend if I get bored.

brice-morin, 2018-01-15:
More than good enough now

closed 2018-01-15 by brice-morin