TelluIoT/ThingML #198: Endianness and binary serialization

ffleurey, 2017-11-05:
The binary serializer plugin we are using only works as long as the serializer and deserializer have the same endianness. Most of the platform we use are little endian (AVR 8bits on arduino, x86 in PCs, ARM on RPi, etc) but for example the MIPS CPU on the Arduino Yun is actually big endian. This makes the serialization between the uC and the MIPS processor wrond with any data type that is more than 1 byte. The way to fix it is not obvious because our binary serialization uses the the ThingML internal binary representation of messages and copies what comes on the serial strait into the scheduler FIFO. Also, when the code is generated in posix C, we do not know what will be the endianness of the target platform.

brice-morin, 2017-11-06:
Well, I guess it is just by chance that the binary serialization for network plugins use the internal serialization. I understand it can be quite optimal to reuse it when targeting little endian. But what prevents you from writing a proper independent serializer (e.g. like it is done for JSON) for big endian? The goal of the serializers is indeed to be able to communicate with non-ThingML system...

brice-morin, 2018-04-18:
I guess the fix for #218 also fixed that one? If not, re-open.

closed 2018-04-18 by brice-morin