TelluIoT/ThingML #199: How to apply OMG's DDS?

Funnjy, 2017-12-05:
Hello,
I'm working on master thesis project where I need to apply DDS with existing CPS system for smart buildings. CSP system's source code is written with ThingML. After compiling program runs as Java application.
Simple version of this CPS receive temperature from one or more thermometers. Based on sensor data system sends signal to turn on/off heater switcher.

I'm currently trying to use RTI Connext DDS but stuck at very beginning. So my question is does anyone experience with RTI Connext/OpenDDS/antother DDS implementation? If so, how/where to start?

brice-morin, 2017-12-05:
Well, is the issue with ThingMl or DDS? Do you have an example of DDS running outside of ThingML? If not, there is not so much we can do to help...

Funnjy, 2017-12-06:
I have simple examples of DDS programs like "Hello World". With RTI connext I can also generate java example of what I need, but again, not sure how to get things working with CPS written on Thingml.

Do you want me to post some code generated by RTI connext?

jakhog, 2017-12-06:
That would be a good start :+1: 

Just to be sure of what you want to accomplish, you want to send/receive messages from ThingML generated Java program, through DDS, to some devices?

Funnjy, 2017-12-06:
@jakhog Yes, something like that. 

"Hello world" example: 
[HelloWorldTypeCodeRTIDDS.zip](https://github.com/TelluIoT/ThingML/files/1536713/HelloWorldTypeCodeRTIDDS.zip)

jakhog, 2017-12-06:
I'm sorry, but we're going to need a bit more specific information about what you actually want to do to help you. Could you describe exactly what you want to do? How do you want ThingML to interact with DDS?

I'm definitely not an expert on DDS either (I didn't know about it before you posted this question yesterday), so I don't fully understand what the program you posted is supposed to do. It seems to me like it contains a HelloWorld message, and Publishers and Subscribers to send and receive it from somewhere...

Funnjy, 2017-12-06:
Ok, I'll try to explain, I'm also newbie with DDS so it is kinda hard :D

DDS is data-centric standard from Object Management Group. 

**What I have today:**
I have CPS designed with thingml. CPS receives data(temperature) from a wireless thermometer. Based on this temperature CPS send a wireless signal to the switcher to turn in on or off. Today CPS is designed to work only with old Telldus gadgets.

**What I need to do:**
I need to apply this data-centric standard for my CPS. It would(in theory) allow me to use whatever swither/thermometer I want without the need to change the source code of CPS. As I understand, a thermometer will be sending(publish?) information to "global data space" where special "topic" exists. CPS will subscribe to this topic in order to receive temperature. Next step is when CPS sends(publish) signal to another "topic" in "global data space". And finally, the switcher is going to receive a signal from that topic in order to react properly on temperature changes(on/off). 

P.s Just let me know if it does not make any sense and I'll try to explain in more details using diagrams.

jakhog, 2017-12-11:
Ok, that I get what you want to do. Looking at the code you sent me, and based on your explanation, it seems like the best way to go is:
1. Make a separate Maven project that exposes an API to send and receive the messages you want from DDS.
2. Include that project in your ThingML CPS code, and make a Thing-proxy that transfers the messages between your other Maven project and ThingML.

When you have a complete Maven project for step 1 - we can provide you with some good examples of how to interface with other Java code from ThingML.

Funnjy, 2018-02-07:
Hello again @jakhog. 

I'm back to my project. Had to fix license for software and it took a long time.

Is there a resource where I can read more about how to create a thing proxy? Same about sending messages between the stomach project with DDS and ThingML project?

closed 2018-06-29 by Funnjy