TelluIoT/ThingML #201: Types library than can be imported from compiler resources

jakhog, 2018-01-15:
We have all gotten used to finding a (hopefully the latest) copy of `_datatypes.thingml` and manually copy-pasting it into our new ThingML project. This is a bit tedious, and unnecessary.

The new parser allows importing stuff from almost anywhere, so one solution is to make a repository of _official_ ThingML libraries/datatypes that can be imported from the web(/GitHub). However, this is not a perfect solution as it requires an internet connection to be able to compile ThingML programs.

A better solution would be to allow importing `.thingml` files that are included in the compilers JAR as resources. I foresee two options for implementing this in practice:
1. Introduce a new syntax for these kinds of imports. E.g. `import <...>` as opposed to `import "..."`
2. Use the existing import syntax, but then when resolving the string to a URI, we first check if the file is present with the given path, and if not, we check if it is present in the JAR resources.

Once this is in place, we also should include a way to copy the _boilerplate_ `.thingml` files from the JAR resources to the project directory (to allow modification by the user). There should be a command in the CLI-compiler, and in the Eclipse UI for this.

brice-morin, 2018-01-15:
For the syntax, maybe `import stl "_datatypes.thingml"` or `import "_datatypes.thingml" from stl` to make explicit what we import from the STL we provide and should maintain. But anyway, syntax is not the hard part here...

brice-morin, 2018-03-22:
@jakhog didn't you implement this?

closed 2018-05-31 by jakhog