TelluIoT/ThingML #208: Bug in Enumeration Type Checking

ffleurey, 2018-02-26:
It seems that the type checker is not able to type Enumeration Literals. It create warnings "parameter cannot be typed" when sending a message with an enum literal as a parameter.

brice-morin, 2018-02-26:
This seems to work (you neeed to put the `@type_checker` on the enumeration itself, not on the literals:

```
import "datatypes.thingml" from stl

enumeration E 
@type_checker "Integer"
{
	FIRST @enum_val "0"
	SECOND @enum_val "1"
}

thing TestCheckFunctionCalls
{
    function foo(a : Integer) : E do
      return E:SECOND
    end
    
    function bar(a : E) : Integer do
    	return E:FIRST
    end

    statechart TestAbstractFunction init INIT {
		//Should fail because foo is called is wrong number of params.
    	state INIT {
    		on entry do
    			foo(E:FIRST)
    			bar(1)
    		end
    	}
    }
}
```

Note however that the typechecker is quite permissive regarding enumeration. If a function return an enumeration (which is typed by Integer), the function is allowed to return whatever integer, including integers out of the range of the enumeration. We could be stricter on enumeration and say that only E:FIRST and E:SECOND (in this example) are valid. But we of course loose some flexibility. But I guess this can be the topic of another issue about the very semantic of Enumeration in ThingML :-)

closed 2018-02-26 by brice-morin

brice-morin, 2018-02-26:
OK, so I also committed a fix in the typechecker itself to be more strict with enumeration. If an enumeration has e.g. `@type_checker "Integer"`, we can use literals as integers, but we cannot use integers as literals. See the two new tests on the checker.

brice-morin, 2018-02-26:
Note also that `@type_checker` on enumeration is now optional. In the previous, if you remove it, it will produce a type error saying that `Function foo is called with an erroneous parameter. Expected Integer, called with E`