TelluIoT/ThingML #210: UML StateChart error when 2 states have the same name in different regions

ffleurey, 2018-03-21:
The mane of the state is used as the ID so it does not go well when we have state with the same name in two different regions (or probably composite state). We should probably use a qualified name to avoid this.

brice-morin, 2018-03-21:
Well, it seems name == ID in plantuml, so that we will have to qualified name as the name for the state (which will show up in the diagrams). We can try to generate the shortest possible qualified that is enough to make sure names are unique. In most cases, it should not change anything (the shortest qname being the name if name is unique).

closed 2018-03-21 by brice-morin

brice-morin, 2018-03-22:
Note that I just use indexes (e.g., MyState_2) instead of qualified names that can rapidly become long

commit c2f1bc65 (2018-03-21): fix #210