TelluIoT/ThingML #213: Bit/byte operators in the action language

brice-morin, 2018-03-27:
A few ThingML programs use bit/byte operators and have to rely on extern expressions, which by chance are the same in all the languages we target so far. Still, this is a hack, not allowing for proper type-checking, etc. We would need to introduce those operators (shift left/righ, bit-wise and, or, xor) in the action language.

jakhog, 2018-03-27:
Just for reference, here are the binary operators that are defined in Java, C, JavaScript and Go:
 - `&` binary AND
 - `|` binary OR
 - `^` binary XOR
 - `<<` binary shift left
 - `>>` binary shift right

brice-morin, 2018-03-27:
Well, do we *really* need this escape mechanism in ThingML? If so, I guess we can change the symbol.

For the `&`, I see the problem.

brice-morin, 2018-11-06:
Well, seems like @jakhog was wrong :-)

- `^` in Go is `~` in all the other supported languages (see [doc for Java](https://docs.oracle.com/javase/tutorial/java/nutsandbolts/op3.html))

That is particularly annoying, meaning that the externs we write so as to deal with those operators are actually not as portable as we thought... i.e. we need, more than ever, to have them in the action language.

@ffleurey comments?

brice-morin, 2018-11-06:
Well, seems like @jakhob maybe was somehow right after all... `^` does exist in Java, but we also use `~` in some places, that work everywhere, but not in Go