TelluIoT/ThingML #214: Go through annotations and see if some should be promoted to 1st class concept

brice-morin, 2018-03-27:
Possible good candidates (feel free to add more):

- `@sync_send`
- `@fork_thread`
- `@type_checker`

We'll create individual issues once we agree on the final list.

jakhog, 2018-03-27:
I don't know if I am supposed to put complaints here, but I have issues with the first two:
 - `@sync_send`. In some languages, we cant do this, e.g. Go. There is no way to do this safely (since they are running in separate goroutines).
 - `@fork_thread`. Some platforms don't have threads (like Arduino). So this wouldn't be that easy. And isn't this kind of why we have `session`s?

brice-morin, 2018-03-27:
Good points. `@sync_send` would also be a pain in Java.

For `@fork_thread` the idea is to allow for simple function-like listeners, which could be useful for external things. Like on entry you could call `listenSerialPort`, which somehow runs in a thread ("physically" or logically) and emits ThingML messages whenever somethings happens on the serial port. Doing that in a session is a bit overkill, as it forces us to have a statechart, etc when all we need is a function that runs in parallel of the rest.