TelluIoT/ThingML #221: Reusing code generated but not yet compiled (and without POM)

nicolasferry, 2018-05-30:
Hey hey,

I have a strange use case and I am not sure this is a good practice but still I am wondering if it is feasible.

I have a set of Java files (a client for SMOOL) that are generated by a tool. These are not compiled and there is nothing like a pom.xml generated. From my side I want to use this code as part of a ThingML model. So I was wondering if there is a way to move these files together with the code generated by ThingML so that they would be compiled together.

Cheers

brice-morin, 2018-05-30:
Hi,

I guess we can define an annotation `@src "path to folder corresponding to the root package"` and copy its content to the src folder of the code we generate from ThingML. I can have a look at that tomorrow. Then all the code should be compiled together. I guess we can even have this path relative, so as if this external code is generate close to the ThingML src, we can have portable path with `../ext/myRootPackage` or similar.

Would that be OK?

Cheers,

Brice.

nicolasferry, 2018-05-30:
Yes, I sounds like it would do it.
Thanks!

commit 4faf0497 (2018-05-30): fix #221 (hopefully)
closed 2018-05-30 by brice-morin